ArduPilot/ardupilot_wiki #2085: QuadPlane Simulation no longer functions as documented

nmcobb, 2019-10-14:
I am attempting to start a QuadPlane simulation, however when running _sim_vehicle.py -j4 -f quadplane --console --map_ as described [here](http://ardupilot.org/plane/docs/quadplane-simulation.html), I am getting a key error: KeyError: 'quadplane'.

Hwurzburg, 2019-10-14:
did you run it from the ArduPlane directory? if not, you need to add -v ArduPlane option

nmcobb, 2019-10-14:
Yeah, I am passing that as well...

Hwurzburg, 2019-10-14:
Runs fine for me

nmcobb, 2019-10-14:
Hmm, do I need to do any setup / compilation? Should I be running off of something other than master?

peterbarker, 2019-10-14:
On Sun, 13 Oct 2019, Miles Cobb wrote:

> I am attempting to start a QuadPlane simulation, however when running sim_vehicle.py -j4 -f quadplane --console --map as described here, I am getting a
> key error: KeyError: 'quadplane'.

Almost certainly a case-sensitivity issue.

closed 2019-10-24 by nmcobb