ArduPilot/ardupilot_wiki #2110: Copter-4.0 wiki changes

rmackay9, 2019-10-20:
a list of changes we should make for the Copter-4.0 release:

- [ ] complex fences (aka stayout zones, inclusion zones, exclusion zones)

Hwurzburg, 2019-10-21:
@rmackay9 first question, does vib fs activate ekf fs or just enable the vib compensation?

Hwurzburg, 2019-10-21:
pre arm messages done...may need a few tweaks....
waiting on answer to question to finish logged error and vib fs updates...

rmackay9, 2019-10-21:
@Hwurzburg, great thanks!  the vibe failsafe only turns on vibe compensation which changes how the EKF calculates it's climb rate (it uses a 3rd order complementary filter) and how the altitude controller calculates the throttle output (it removes the lowest level acceleration controller).

rmackay9, 2019-10-22:
@Hwurzburg, as discussed here are the important bullet points about the Copter vibration failsafe:

Why is it important?

- Multicopters can suffer from very high vibration levels (more 60m/s/s) which can lead to the accelerometers saturating (i.e. going beyond the range that the sensor can measure).  This leads to "clipping" and means the EKF cannot accurately calculate it's climb rate or vertical acceleration.  This can lead to the vehicle becoming unable to control it's climb rate and in severe situations can lead to the vehicle climbing rapidly at full throttle.

When will it trigger?

- when the EKF's vertical velocity and position innovations becomes negative AND the velocity variance is less than 1... and this situation persists for 1 second

Note: 
- An Innovation is the difference between the predicted value and the latest (non-IMU) value.
- A Variance is the EKF’s reported confidence in its estimate.  0 is very good, >1 is bad

What will happen when the failsafe triggers?

- "Vibration compensation ON" will appear on the Ground Stations HUD.
- EKF's climb rate will be calculated using a 3rd order complementary filter which is tuned to be more resistant to vibration (but less accurate) than the normal method
- Altitude controller switches to a more vibration resistant two-stage controller (position->velocity) instead of the regular three-stage controller (i.e. position->velocity->acceleration)
- The vehicle will not change mode but it's altitude hold will be less accurate than normal.  The vehicle may overshoot it's altitude targets and/or respond more slowly to pilot input.

When will the failsafe clear?
- the vibration compensation will stay on for at least 15 seconds.
- when/if the EKF recovers, "Vibration compensation OFF" will be displayed on the HUD and the climb rate and altitude controllers will return to the regular method.

How to disable the failsafe:

- set FS_VIBE_ENABLE = 0

Hwurzburg, 2019-10-26:
I took the release notes and updated first list