ArduPilot/ardupilot_wiki #2119: Contradictory advice on (non)use of TX trims

kd0aij, 2019-10-22:
Here: http://ardupilot.org/plane/docs/fpv-plane.html?highlight=servo_auto_trim
we say "NEVER adjust your TX trims or subtrims again, all further trim procederes will be performed on your FC exclusively!"
meaning that **TX trims are always zero**.

but here: http://ardupilot.org/plane/docs/fixed-wing-faq.html#why-does-my-trim-change-when-i-change-modes
we say "setup the plane with correct trims in MANUAL by adjusting using the trip tabs in flight"
implying that **TX trim is non-zero and the same for all flight modes**.

Hwurzburg, 2019-10-22:
Thanks...I have fixed it, as well as updating the page to new standards (every param is to be hyperlinked), but cant build and test, since the main wiki site is broken and cant build, and also locally cant pull the params for link testing on a local build....once everything is back up, will test and will merge the fix...

Hwurzburg, 2019-10-25:
fixed....closing this

closed 2019-10-25 by Hwurzburg