ArduPilot/ardupilot_wiki #2132: "Forum" link on left menu bar of index page

rmackay9, 2019-10-24:
I'm not sure if it's new or old but I noticed a "Forum" link is on the left side of the ardupilot wiki (http://ardupilot.org/ardupilot/).  Perhaps i'm being picky instead of practical but I think this breaks the underlying rule that external links are in the upper menu and internal links are on the left bar.  Of course, if @Hwurzburg prefers it this way then I'm happy with it.

As a side note I think that AntennaTracker and Developer should be moved up under "Sub" and above "Firmware Downloads".

I also wonder if we should archive "APM Planner 2".  Perhaps it needs a team discussion but I think it's not being maintained.

auturgy, 2019-10-24:
Last commit to apm planner was only a month ago: not hugely active, but ticking along.

Regards,

James

Hwurzburg, 2019-10-24:
Re-ordered menu to match top menu for sites....removed forum link

rmackay9, 2019-10-25:
I think @auturgy is right that we can leave APM Planner 2 because there is still activity there.

closed 2019-10-25 by rmackay9