ArduPilot/ardupilot_wiki #2162: Plane-4.0 Wiki Updates Needed

Hwurzburg, 2019-10-28:
Looking at release notes, the following items need work:
- [ ]QAUTOTUNE mode - needs adding , I need to understand how it works
- [ ]new simulators: Morse, Webots, Scrimmage: Addition to SITL in Dev
- [ ]added AFS_MAX_RANGE for limiting vehicle range:update AFS page
- [ ]added support for reversible blheli32 ESCs: needs expansion in AutoLanding addition as advanced configuration
- [ ]support DLVR I2C airspeed sensors: Arspd sensor page needs updating for type, also note on not using sensor

Hwurzburg, 2019-10-28:
@tridge  I'll need some classroom time with you...

Hwurzburg, 2019-10-30:
@tridge cant seem to find a Q_AUTOTUNE_AXES parameter....dont think it exists in the code...

rmackay9, 2019-10-31:
@Hwurzburg, the Q_AUTOTUNE_AXES parameter is hidden down in [AC_AutoTune library here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AutoTune/AC_AutoTune.cpp#L105).  I've pulled up QuadPlane in MP's simulator and after setting Q_ENABLE = 1 I can see the parameter.  It doesn't seem to have a parameter description though so perhaps plane has some issue with a reference to the params but this is probably out-of-scope.

Hwurzburg, 2019-10-31:
@tridge @rmackay9 well if its not in the full parameter list, then it cant be referenced with a link in the wiki....other "hidden" params like OSD /Batt/RSSI params that arent visible until enabled, but are still listed in the parameters list and pulled down to the wiki for referencing...I think that we need to find out the reason they are not being included and fix it....cant really finish the AUTOTUNE mode page without that param and its ready for a PR except for that...

rmackay9, 2019-11-01:
@Hwurzburg, sure I'll have a look..

Hwurzburg, 2019-11-01:
@rmackay9  .... @tridge already made a change to fix it...and they all now appear...

Hwurzburg, 2020-05-15:
completed, closing

closed 2020-05-15 by Hwurzburg