ArduPilot/ardupilot_wiki #2199: Missing video of biwing quad tailsitter

peterbarker, 2019-11-06:
https://youtu.be/cfqP9-2IWtQ

Differential thrust for roll/pitch.

Also has control surfaces.

http://ardupilot.org/copter/docs/common-all-vehicle-types.html

peterbarker, 2019-11-06:
Also, perhaps this one deserves an entry?

https://www.youtube.com/watch?v=ed6Et6ENU-g&feature=youtu.be

peterbarker, 2019-11-06:
This is a X vs + configuration for the motors.

kd0aij, 2019-11-07:
copter tailsitters (quad+) by Pierre Losa
jetwing: https://www.youtube.com/watch?v=TtF8WSfgQhI
https://www.youtube.com/watch?v=wMVBBoANOyw
hacker wing: https://www.youtube.com/watch?v=ed6Et6ENU-g&t=2s

kd0aij, 2019-11-07:
It appears that the only link to "all supported vehicles" is in the copter sub-wiki, yet it includes quadplanes and rovers?

Hwurzburg, 2019-11-07:
no its in every vehicle...just not  obvious...its on each intro page as a link

kd0aij, 2019-11-07:
maybe it would be better to have a link below "Home" on this page: http://ardupilot.org/ardupilot/

Hwurzburg, 2019-11-07:
certainly could be added there....even  restricted once in a sub-wiki from their intro pages....and maybe should be on the home subwiki page left menu also...

kd0aij, 2019-11-07:
This is a SITL demo of the "manual forward throttle" PR for quadplanes: https://youtu.be/QDexNUvDX0I

Hwurzburg, 2019-11-07:
#2200

Hwurzburg, 2019-11-07:
maybe on the Ardupilot channel...have any others besides this or from others ? otherwise if you want feedback, post it on the blog?

closed 2019-11-07 by Hwurzburg