ArduPilot/ardupilot_wiki #2203: move watchdog to be under failsafes?

rmackay9, 2019-11-08:
I think we should move the new [Independent Watchdog](http://ardupilot.org/copter/docs/common-watchdog.html) page to be under [Failsafes at least for copter](http://ardupilot.org/copter/docs/failsafe-landing-page.html).

Some other minor changes:

- I think the independent watchdog is on by default so it can be **disabled** by setting BRD_OPTIONS = 0.
- Let's say "ArduPilot 4.0 and later firmware.."

I think we should include some info on how someone can recognise that a reset has occurred.  One way is by looking at the dataflash logs.  If the log is filtered to show only the "MSG" messages we can see some that include the word, "watchdog".  This is a clear indication that the **previous** log or flight ended with a watchdog reset.
[image]

Hwurzburg, 2019-11-08:
I can do that for copter/rover...but Plane has to stay as Advanced config since its not organized like Copter on failsafes...I have a PR to reorg Plane some, once that is in , I can continue to reorg its failsafes also and then move IWDG to that new landing page...

rmackay9, 2019-11-08:
Thank very much!

Hwurzburg, 2019-11-08:
didnt realize rover doesnt have the same landing page as copter....so my plan is to leave it in advanced config for all vehicles to assure TOC entries, add it to copter failsafe landing page and mention in the failsafe pages of plane and rover...until they get reorg'd to be more copter-ish

rmackay9, 2019-11-08:
Yup, great, thanks!

closed 2019-11-08 by rmackay9