ArduPilot/ardupilot_wiki #2248: [Update on ]Communicating with ODroid via MAVLink

piyushchauhan, 2019-11-21:
[Communicating with ODroid via MAVLink](http://ardupilot.org/dev/docs/odroid-via-mavlink.html) - 
Update the links of harwares and others parts. There directed to 404 page.

brunoolivieri, 2019-11-21:
@piyushchauhan thanks for the information. Please, could you propose the updates?

piyushchauhan, 2019-11-21:
Will do soon.

Hwurzburg, 2026-08-28:
Closing this one, as archiving has since answered it.

The page is now **Archived: Communicating with ODroid via MAVLink**, listed under [Archived Topics](https://ardupilot.org/dev/docs/common-archived-topics.html), whose stated purpose is "topics that are no longer considered relevant to most users, but which may have some value to users with old hardware."

The hardware is the reason. The U3 was already marked discontinued when this issue was filed, and the C1+ and XU4 have since left Hardkernel's lineup too. Of the 12 hardkernel.com links on the page, 9 now return 404 and 3 still resolve - and refreshing them is not possible, because the products they pointed at no longer exist to link to.

Rather than keep a parts list pointing at a vendor catalogue that has moved on, the page stays archived as a historical record for anyone still running one of these boards. Dead shop links on an archived page for discontinued hardware are expected rather than a defect.

Feel free to reopen if you feel this has not been addressed.

closed 2026-08-28 by Hwurzburg