ArduPilot/ardupilot_wiki #2261: Copter Object Avoidance

auturgy, 2019-11-25:
[Copter Object Avoidance](http://ardupilot.org/dev/docs/code-overview-object-avoidance.html) - Dev page is out of date with setup page for avoidance (missing bendy ruler and dykstra's - http://ardupilot.org/copter/docs/common-object-avoidance-landing-page.html)

Hwurzburg, 2019-11-25:
@peterbarker whom should this be assigned to? I dont have the requisite knowledge to address these OAs in terms of Dev perspective...

peterbarker, 2019-11-26:
In this case, @rmackay9 would be appropriate.  I'd have to research to get things right (I can usually point out things that are incorrect, but not add content).

One option you have for determining appropriate people to poke is to use `git log` and `git blame` on the relevant source files to see who has worked on the files.

Additionally, of course, the fact that this mentions the vehicle type is a fair indication the vehicle maintainer needs to be involved in some way :-)

rmackay9, 2020-01-29:
We've added the bendyruler and dijkstras pages now so closing

closed 2020-01-29 by rmackay9