ArduPilot/ardupilot_wiki #2276: Let's Encrypt SSL certificate on website

tenten8401, 2019-11-28:
I don't know if this is the right repository, but it'd be extremely nice to have SSL support on the main website, especially since most browsers now have a "not secure" badge next to the URL. https://letsencrypt.org/ gives out certificates for free, is completely automated and takes like 5 minutes to set up.

I know we aren't really transferring anything sensitive, but it'd be nice to have to prevent MiTM attacks among other things.

khancyr, 2019-11-28:
Hello, 

That something we discuss from some time already but that have, to my knowledge, more implications than just use let's encrypt : SEO, wildcard support, etc.
Anyways, some of us are already working on this. It should come fast.

Thanks for pointing it out !

auturgy, 2019-12-07:
@tridge i think we can close this

jpkh, 2019-12-07:
I can add certs on wiki. We are using LE certs on many of our other websites but not  in wiki

auturgy, 2020-05-07:
LE certs have been implemented.

closed 2020-05-07 by auturgy