ArduPilot/ardupilot_wiki #2283: Boogie Board Boat

adalta87, 2019-11-29:
[Boogie Board Boat](http://ardupilot.org/rover/docs/reference-frames-boogieboard-boat.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I have created some DXF, sketchup, and STL files for the thruster mounts shown, DXF should work for anyone wanting to get them CNC from a local shop, the sketchup file should allow anyone that has it to change it and the stl would allow someone to print it, its 3mm tall. Would you like them to add to the wiki?

rmackay9, 2019-11-30:
@adalta87, sure, that sounds good, thanks!

I've been meaning to add my parameter file to the page as well... I'll do that soon I hope.

adalta87, 2019-11-30:
Can I just send them to you via here. I tried to upload them in my comment but not allowed format. Only difference on the motor mount i made is the corners aren't rounded on the outside corners which i can easily make one like that. 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ArduPilot/ardupilot_wiki/issues/2283?email_source=notifications\u0026email_token=AFODXT5JNRXDHQYBRJAL7G3QWGWJTA5CNFSM4JTD5OBKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFPV7PI#issuecomment-559898557",
"url": "https://github.com/ArduPilot/ardupilot_wiki/issues/2283?email_source=notifications\u0026email_token=AFODXT5JNRXDHQYBRJAL7G3QWGWJTA5CNFSM4JTD5OBKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFPV7PI#issuecomment-559898557",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

rmackay9, 2019-11-30:
@adalta87, sure, maybe put them somewhere (like dropbox or equivalent) and provide a link ?

adalta87, 2019-11-30:
https://drive.google.com/open?id=1Oa_L0CaEaoasALAftZQOwm9mhO2C7uld

let me know if that works. I will add files to that folder for the one I am
making.  Here are some interesting thingiverse files folks can print could
also link them.

https://www.thingiverse.com/thing:1497003 More reinforced mount for motor
https://www.thingiverse.com/thing:1498338  Motor Guard