ArduPilot/ardupilot_wiki #2285: add "Trusted, Versatile, Open" to logos on wiki

tridge, 2019-12-01:
We should have "Trusted, Versatile, Open" in several places in our wiki. Key places are:

 - under the 'ArduPilot' logo on ardupilot.org
 - under the ArduPilot logo in left bar on http://ardupilot.org/ardupilot/
 - under ArduPilot logo on each vehicle page

It has been our motto for a long time, but we've never pushed it out so is not widely known

Hwurzburg, 2019-12-01:
cant seem to find an ArduPilot logo on a vehicle page, index nor intro..can you point it out?
we should assign this to whomever can edit the concrete

tridge, 2019-12-02:
I think we should have "ArduPilot" with "trusted, versatile, open" at least on front page of every vehicle wiki, at top of left hand menu. Perhaps on every page that displays that left colum menu?

tridge, 2019-12-02:
we should also start the process of removing the big "ArduPlane" and similar images, and use ArduPilot consistently

auturgy, 2019-12-07:
Duplicate of https://github.com/ArduPilot/ardupilot_wiki/issues/1091
They can probably both be closed: thanks @Hwurzburg

closed 2019-12-07 by Hwurzburg