ArduPilot/ardupilot_wiki #2294: Add info on ICE start stop during vtol takeoff and land

Hwurzburg, 2019-12-03:


Hwurzburg, 2019-12-19:
@tridge ...need a couple of bullets on what happens/controls this...then will write up a PR...thanks

tridge, 2019-12-24:
we have a DO_ENGINE_CONTROL command that can be added before a NAV_VTOL_TAKEOFF. It takes 3 parameters:
  param1: start/stop. Set to 1 to start engine, 0 to stop
  param2: cold start (1 for using choke, currently not implemented)
  param3: height delay in cm. This gives the height at which engine will start

for landing there is a parameter Q_LAND_ICE_CUT. If this is set to 1 then the engine will stop when you are in the final stage of vtol landing.

Both of these options rely on the ICE subsystem being enabled

Hwurzburg, 2019-12-25:
does this apply to copter and rover also?

Hwurzburg, 2020-01-09:
added

closed 2020-01-09 by Hwurzburg