ArduPilot/ardupilot_wiki #2300: add "Air Mode" reference

tridge, 2019-12-05:
One of the common Qs for people coming from the betaflight world is how to do "Air Mode" in ArduCopter. "Air mode" in betaflight means that stabilisation is active at all throttle levels in acro and stabilize mode, so if you go to zero throttle the vehicle keeps stabilising. It is a binary option in betaflight, and very widely used.
We do have it in ArduCopter, but it is not called "Air mode". We should make sure a search finds an explanation of it for users used to betaflight.
In ArduPilot you enable it by enabling arming on a RCn_OPTION switch. If you arm with a switch then we don't disable stabilisation at zero throttle, the theory being that if you want the motors off you can flick the arming switch.
This is important both for racing, and for tricks like loops, freestyle etc

rmackay9, 2019-12-05:
OK, maybe we can add a note to the auxiliary switch page: http://ardupilot.org/copter/docs/common-auxiliary-functions.html

The current description is, "Arms the vehicle if the switch goes high (subject to arming checks). Disarms the vehicle if brought low."

Let's add something like this to the end, "In copter, this option has the side effect that the vehicle will continue to stabilize in Acro and Stabilize mode when the input throttle is at zero.  This is called "AirMode" in BetaFlight"

peterbarker, 2019-12-05:
On Wed, 4 Dec 2019, Andrew Tridgell wrote:

> In ArduPilot you enable it by enabling arming on a RCn_OPTION switch. If you
> arm with a switch then we don't disable stabilisation at zero throttle, the
> theory being that if you want the motors off you can flick the arming
> switch.

@tridge said two things here.  The second is correct.  It isn't sufficient 
to just enable arming on an RCn_OPTION switch, you *also* have to actually 
use it for arming.  So if you have it on a switch but then use 
rudder-arming then you won't get "airmode".

rmackay9, 2019-12-05:
We could also add notes to the Stabilize and Acro pages.

On the [stabilize page's "Overview" section](http://ardupilot.org/copter/docs/stabilize-mode.html#overview) it says this:

> Pilot’s throttle input controls the average motor speed meaning that constant adjustment of the throttle is required to maintain altitude. If the pilot puts the throttle completely down the motors will go to their minimum rate (MOT_SPIN_ARMED) and if the vehicle is flying it will lose attitude control and tumble.

we could add "unless an auxiliary switch is setup to Arm/Disarm disarm the vehicle" and the words "auxiliary switch" could link to this page: http://ardupilot.org/copter/docs/common-auxiliary-functions.html

[Acro mode](The throttle is completely manual with no compensation for tilt angle of the vehicle. If the pilot puts the throttle completely down the motors will go to their minimum rate.) has a similar overview section but the throttle explanation is shorter.  I think we could just make it the same as for stabilize.

rmackay9, 2019-12-05:
@peterbarker, oh really?  that's tricky.  I guess that behaviour is OK .. it could be argued either way.. but it's tricky.

tridge, 2019-12-05:
a key thing in this issue is to use the words "Air mode" somewhere so a google search of "ArduPilot air mode" finds the page

Hwurzburg, 2019-12-05:
Even thought its not a "mode", I would list it as such on the flight modes page with the appropriate words on the linked description page stating its not a true mode,etc.......this is a huge feature for mini-quad flyers and warrants easy, stand-out visibility in wiki...I wasnt aware it was even a capability...

rmackay9, 2019-12-05:
You could also put it just above "Boat Mode" in the Advanced section. http://ardupilot.org/copter/docs/boat-mode.html

Boat mode is also not a mode but people kept calling it that so we added the page.

I leave it up to you where to put it though..

Hwurzburg, 2019-12-14:
#2304

Hwurzburg, 2019-12-19:
done

closed 2019-12-19 by Hwurzburg