ArduPilot/ardupilot_wiki #2309: ROS and Hector SLAM for Non-GPS Navigation

kenhuyang, 2019-12-07:
link: https://ardupilot.org/dev/docs/ros-slam.html
the article misses some details on how to transmit the slam data to ardupilot. the Hector SLAM publishes "/slam_out_pose" and "/poseupdate" for the final result, but mavros need  "/mavros/vision_pose/pose" and "/mavros/vision_pose/pose_cov" topic, so we need to add the following two lines at $HOME/catkin_ws/src/hector_slam/hector_slam_launch/launch/tutorial.launch
  <remap from="/slam_out_pose" to="/mavros/vision_pose/pose" />
  <remap from="/poseupdate" to="/mavros/vision_pose/pose_cov" />"

kenhuyang, 2019-12-07:
<remap from="/slam_out_pose" to="/mavros/vision_pose/pose" />
  <remap from="/poseupdate" to="/mavros/vision_pose/pose_cov" />

Omiclous, 2023-09-26:
Hi, Did this work? I've been trying, but these topics don't remap for me