ArduPilot/ardupilot_wiki #2318: Wheel Encoders

adalta87, 2019-12-09:
Please change the ground testing from MP status screen RPM 1 and RPM 2 to make sure wheel encoders are working please Crtl-F within MP and select mavlink inspector, scroll to the bottom of the status messages to wheel_distance. The values will change if wheel encoders are working correctly.
[image]

rmackay9, 2019-12-09:
Yes, I think this is a good change to make, 4.0 users should check the WHEEL_DISTANCE.distance vlaues instead of RPM.

Hwurzburg, 2019-12-24:
done...closing

closed 2019-12-24 by Hwurzburg

adalta87, 2019-12-24:
henry your the shit! You shouldn't be working on Christmas eve hahahah!
Happy Holidays!