ArduPilot/ardupilot_wiki #2321: Links on MP antenna tracker page are broken

rmackay9, 2019-12-10:
The links to the images on this page result in a 404 error https://ardupilot.org/copter/docs/common-mission-planner-gps-based-antenna-tracking.html

This was brought up in this discussion: https://discuss.ardupilot.org/t/photos-missing-in-build-log/50070/1

Hwurzburg, 2019-12-11:
images are being hosted off site....therefore they are not copied from source to site's _images folder...the image hot links never worked I believe....

two approaches....remove the hot links, quick and easy....
copy and move all offsite images into source images....more involved

not having studied the tracker issue at all, my question is: does this article even have significant merit any more with the advent of AP antenna tracker....if so, we need to do latter time consuming approach....if not, then the first fix is quick and easy...

in addition, this page is not secure for some reason.....@brunoolivieri @tridge  any idea why an internal page is not secure? perhaps since it references offsite images?

brunoolivieri, 2019-12-11:
Yes, the page is not secure because it has mixed content (the images from the external site). 

I believe that the images cames from here. I do not think we could have copyrights issues by internalizing them. If you agree, I suggest inserting all the photos in our Images folder because it would fix both problems. I did the PR #2323 with this approach.

IFF copyrights could be an issue, we could only replace the targets to the image link itself and stay with a security warning.

Hwurzburg, 2019-12-11:
fix merged...thanks Bruno!

closed 2019-12-11 by Hwurzburg