ArduPilot/ardupilot_wiki #2335: Toshiba CAN ESCs

WondesenB, 2019-12-17:
Hello,
When I powered Toshiba ESC it continuously flashing with orange LED. On the manual it says  "2s_ON/ 1 s_OFF" to solve this problem.  I couldn't able to solve this problem. Any help will be appreciated !
Thanks !

rmackay9, 2019-12-17:
@WondesenB, well, this is the very first support call we've received re the ToshibaCAN ESCs!

Would it be OK if we tested using Copter-4.0.0-rc3 (the latest beta)?  Also instead of doing the support here on the wiki to-do list it would be more convenient to do it in the copter-4.0 support forum: https://discuss.ardupilot.org/c/arducopter/copter-40 so could you follow this link and create a new discussion?

I'll close this issue but I'll monitor it for replies in case you have troubles creating a new discussion.

closed 2019-12-17 by rmackay9

WondesenB, 2019-12-17:
Accepted ! Thank you  !

በሰኞ፣ ዲሴም 16 2019 11:20 ከሰዓት ላይ Randy Mackay <notifications@github.com> ላይ
የሚከተለውን ጽፈዋል፦

> Closed #2335 <https://github.com/ArduPilot/ardupilot_wiki/issues/2335>.