ArduPilot/ardupilot_wiki #2358: add note on auto servo range for DShot

tridge, 2019-12-24:
when we setup motors using the SERVO_BLHELI options we automatically force output channels setup using SERVO_BLHELI options to use a range of 1000 to 2000. This can confuse users so we need to document it.
See this discussion:
https://discuss.ardupilot.org/t/plane-4-0-release/47188/205
and this issue:
https://github.com/ArduPilot/ardupilot/issues/13092
Note that we force the trim to either 1000 or 1500 depending on whether the output is marked as reversible.

Hwurzburg, 2019-12-25:
done

closed 2019-12-25 by Hwurzburg

Rolf-G, 2019-12-26:
BLHELI-telemetry in non-vectored tri-motor VTOLs works with SERVO_BLH_AUTO set 0 and instead SERVO_BLH_MASK with a bitmask of the actually outputs connected to BLHELI-ESCs.

Rolf-G, 2019-12-26:
@Hwurzburg  ,  @tridge 
as a result of the above, i suggest adding the following section in the wiki on page 
https://ardupilot.org/plane/docs/guide-tilt-rotor.html 
after the section "Vectored Yaw" 

> Non-Vectored Yaw
> ================
> 
> Non-Vectored yaw aircraft (Q_TILT_TYPE 0 or 1) needs a tilt servo for the rear
> motor for yaw control.
> 
> You need to setup your two front tilt servos both with SERVOn_FUNCTION=41
> and also your rear servo for yaw control with SERVOn_FUNCTION=39 (no literal error).
> 
> Note that if you want to use BLHELI ESCs with telemetry in non-vectored tri-tilts, 
> you must not set BLHELI to 1. 
> Set SERVO_BLH_MASK to the output-bitmask of the servo-channels actually connected BLHELI-ESCs. 
> E.g. SERVO_BLH_MASK=1792 for Servo 9,10,11 (the first three aux-outputs of a pixhawk1). 
> You also need to setup Q_M_PWM_TYPE to 4 (DShot 150).
> 

-----------------------------------------