ArduPilot/ardupilot_wiki #2366: Release Notes Wiki Page

Naterater, 2019-12-26:
https://ardupilot.org/plane/docs/project-news.html, having the link to the github page with the release notes is an excellent start for a uniform location for the release notes. To take this to the final step, I think a few options could be considered.

1. Create or rename the "project-news" html to be "releases" or "release-notes" so that the page title and URL match and there is no confusion. Maybe even redirect to a simpler URL like ardupilot.org/plane/releases
2. Advertise this page on the ardupilot.org homepage or somewhere higher in the left-hand menu/tree instead of at the bottom in the appendix section.
3. Auto-generate the text and bullet points on the actual wiki page to avoid redirecting to the github format (that most non-code users are generally uncomfortable with). This doesn't need to be physically copying and pasting the same information; leave the github page as the original source and the wiki page should reference it. This is done with the full parameter list, so I'm assuming it can be done with this page as well. It should only need to be updated once daily.
4. Add button(s) to subscribe to mailing list(s) that will inform them of new stable or beta releases of firmware by email. I think people have been asking for this as the social media and forum posts can be buried quickly in just a few days' time.

stephendade, 2020-04-19:
As an update to this, @auturgy and myself are working on a method to manage/release safety bulletins. We gave an initial talk about at at the 2020 Developer's Conference: https://www.youtube.com/watch?v=w6dwC9udmjE&list=PLC8WVaJJhN4xGtB7TDT5OQTeOIkhzTrcA&index=2

Hwurzburg, 2020-09-06:
User notification is now active....can we close this now?