ArduPilot/ardupilot_wiki #2368: Error downloading firmware on Missionplanner.

DroneFletch, 2019-12-27:
When I try to get the firmware for my arducopter board the following error appears...
Failed to download new firmware : System.UriFormatException: Invalid URI: The URI is empty.
at System.Uri.CreateThis(String requestUriString)
at MissionPlanner.Utilities.Firmware.updates(String comport, software temp, String historyhash)

Then another tab comes up immediately after saying error uploading.

auturgy, 2019-12-27:
Can you provide more details please: your board type and Missionplanner version would be helpful.

Regards,

James

DroneFletch, 2019-12-27:
I believe I have a 2.6. I tried downloading the firmware from APM planner software and that was successful.

auturgy, 2019-12-27:
Ah ok. I’m not sure how Missionplanner support is for those, but the firmware hasn’t supported them for a number of years, so they run very old code. Good luck!
FYI currently supported boards are listed here: 

https://ardupilot.org/copter/docs/common-autopilots.html

Regards,

James

Hwurzburg, 2019-12-27:
@DroneFletch As @auturgy states APM2.6 is not supported any longer. Many supported boards are very inexpensive now, it would probably be worthwhile upgrading.
FYI if you need support in the future, try posting to discuss.ardupliot.org,  as this is for Wiki documentation issues....thanks!

closed 2019-12-27 by Hwurzburg