ArduPilot/ardupilot_wiki #2388: Add Autorotate to flight mode page

rmackay9, 2020-01-01:
We should add Heli's autorotate flight mode to this page: https://ardupilot.org/copter/docs/flight-modes.html because at least one Copter-4.0.x user is asking about it.

This user question came up here on the forums: https://discuss.ardupilot.org/t/autorotate-systemid/50624

As a side note I think we should consider modifying the FLTMODE1,2,etc parameter descriptions so that it's more clear that it's heli only.  Perhaps change "26:Autorotate" to "26:Autorotate (Heli only)"

@Gone4Dirt, @bnsgeyer FYI

Hwurzburg, 2020-01-01:
no params associated with this mode? anything we should explain about how it works?...every mode should have its own page...fixing the FLTMODE params is easy, as is adding it to the wiki modes list, but it needs an explanatory page about how its used...the phases, what user inputs are allowed/needed in each phase...having auto'd nitros and electrics hundreds of times in the past (Ron Lund and I traded lots of $$ in "autoing for dollars" contests), I should be able to write something up with just some bullets....

Gone4Dirt, 2020-01-01:
I am already working on this.  I am writing the flight mode page as we speak.  I will add it to the flight modes page that Randy linked and link the explanation page when it's available.  The PR for the FLTMODE param meta data is already in.

Gone4Dirt, 2020-01-02:
PR for the wiki flight mode is now in.

Hwurzburg, 2020-01-03:
closing since there is an active PR

closed 2020-01-03 by Hwurzburg