ArduPilot/ardupilot_wiki #2397: RoadMap for 2019/2020 - non-gps-navigation  - T265 COMPLETED

patrickpoirier51, 2020-01-03:
@rmackay9  

https://ardupilot.org/dev/docs/roadmap.html#non-gps-navigation-avoidance-and-path-planning-randy-patrick-poirier

I think we can consider this section as COMPLETED
Add support for Intel RealSense T265, D435 (ROS & non-ROS)

As for SLAM improvement, as I wrote on the dev funds mailing list, further developments depends on an updated  EKF that would permit a better integration from external estimators.

rmackay9, 2020-01-04:
@patrickpoirier51, yes, I agree.  I'll update the roadmap wiki page (unless someone beats me to it).

Thanks for the input on the funding list as well.  I'll send out a vote shortly and let's see how it goes..

auturgy, 2020-05-07:
Page was updated

closed 2020-05-07 by auturgy