ArduPilot/ardupilot_wiki #2410: Holybro Kakute F7 wrong orinetation

mafrmt00, 2020-01-06:
[Holybro Kakute F7 and KAKUTE F7 AIO](https://ardupilot.org/copter/docs/common-holybro-kakutef7aio.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

rmackay9, 2020-01-07:
Thanks for the report.  This should really go onto the main flight code issues list: https://github.com/ArduPilot/ardupilot/issues but in any case, I'm surprised this hasn't come up before.  I think we have numerous KakuteF7 users including @andyp1per.

andyp1per, 2020-01-07:
@mafrmt00 lot's of people using this without issues. Can you specify which exact board you are using? How are you determining that left and right are swapped? The OSD horizon should move in the opposite direction to your roll

mafrmt00, 2020-01-07:
Sorry,misinterpreted the OSD.

closed 2020-01-07 by mafrmt00