ArduPilot/ardupilot_wiki #2423: LEDs: split external LEDs into two pages (or maybe three if we want a landing page)

rmackay9, 2020-01-10:
I think we should split the External LEDs page into a landing page and then under this have separate ToshibaLED and NeoPixel LED pages.

The landing page should perhaps keep a link to the other wiki page that describes the meaning of the lights and perhaps there's other general information about LEDs that could also remain on the landing page.

Hwurzburg, 2020-01-10:
I would need info on all the LED types, since there are many more than Toshiba and Neopixel supported, and how each one is connected.....who can explain that to me?

Hwurzburg, 2020-01-16:
#2423 addresses this, I hope...but #2424 must be merged first since it has an image required...
rather than splitting into multiple pages now, I think this is sufficient, I think that we need to also update the servo outputs page and I have put it on my list...

Hwurzburg, 2020-01-22:
@rmackay9 okay to close?

closed 2020-01-28 by Hwurzburg