ArduPilot/ardupilot_wiki #2430: Add Zubax setup information

rmackay9, 2020-01-13:
Our [zubax setup wiki page](https://ardupilot.org/copter/docs/common-zubax-gnss-positioning-module-gps-compass-and-barometer.html) re-directs users to the Zubax site with the message, "The Zubax website has all the documentation needed to use Zubax GNSS 2 with ArduPilot" but when I followed the links, I don't see any ArduPilot setup information actually.

I think we should add the setup information to our page and then simply say that the Zubax setup page is "here" as a reference (and provide the link)

auturgy, 2020-01-13:
Pavel removed the ArduPilot documentation from the Zubax website.
See
https://forum.zubax.com/t/gnss2-don-t-work-with-arduplane-3-9-0/1054/6

Regards,

James

Hwurzburg, 2020-01-13:
given that they are not willing to put any effort themselves into the issue, I recommend that we archive the page until we get a unit and debug the UAVCAN issue it has with ArduPilot

rmackay9, 2020-01-13:
For the sake of the AP users we should be able to document how it works with AP and I'm very sure it does work because I think @tridge has one.

I think all the user needs to do is just to set these two parameters and it should work:

- CAN_P1_DRIVER = 1 (to enable CAN)
- GPS_TYPE = 9 (UAVCAN)

Hwurzburg, 2020-01-13:
also a few other things, but they are documented in the UAVCAN setup page in the wiki....but I think he is complaining that it does not plug and play? and then Zubax says "we will pull our docs and recommendation" since they dont??? since we cant see the docs, we dont know if they are right or wrong, whether the user is not following them or what...
if someone has one, and can confirm it works, I can add links on the page to our UAVCAN docs...it should work

Hwurzburg, 2020-01-13:
#2434

daveglover90, 2020-01-13:
Mine should be coming in on Wednesday or Thursday.  I will give it a try.  

I also just bought the CUAV Neo V2 pro which is can based as well.  Will try that out too

I will report back as soon as I try then out

rmackay9, 2020-01-15:
I think we've fixed up the wiki now so closing, thanks!

closed 2020-01-15 by rmackay9