ArduPilot/ardupilot_wiki #2431: RangeFinder Setup Overview should specify PRX_TYPE = 4

rmackay9, 2020-01-13:
The [RangeFinder Setup Overview page](https://ardupilot.org/rover/docs/common-rangefinder-setup.html) discusses how to setup 8 horizontally facing lidar for use in Object Avoidance but we should also mention that if 8 individual rangefinders are setup pointing horizontally that for them to be used for object avoidance the PRX_TYPE = 4 param must also be set.

Hwurzburg, 2020-01-13:
if you follow the link in that sentence to Object Avoidance, which requires setup for those rangefinders to be used, it leads you to the avoidance use pages and that value is shown several times...you think it needs duplication?

JosepBC, 2021-03-13:
Hello, I've had some troubles setting up the configuration for object avoidance since it doesn't say that you should use PRX_TYPE=4 in order to use object avoidance in AltHold mode as well.
I understood that it was needed just in Loiter mode because the param is under Loiter.

rmackay9, 2021-03-17:
I agree we should add the PRX_TYPE=4 on the RangeFinder Setup Overview page.

Rangefinders are used for all kinds of things including terrain following, optical flow, precision landing and the three types of obstacle avoidance.  The page mostly talks about object avoidance but misses the important info that PRX_TYPE should be set to 4.