ArduPilot/ardupilot_wiki #2447: Scripting additions that are not documented:

WickedShell, 2020-01-15:
This is a general tracking list for things that have come into master but didn't get mirrored into documentation yet. I will eventually get to it, or someone else can beat me to it, but I wanted to maintain a list of the features here so it would be a bit easier to figure out what to add docs for.

- [ ] Battery cycle count - https://github.com/ArduPilot/ardupilot/pull/13035
- [ ] Vehicle likely flying, and baro calls - https://github.com/ArduPilot/ardupilot/pull/13118
- [ ] Vehicle get mode - https://github.com/ArduPilot/ardupilot/pull/13213
- [ ] UART drivers - https://github.com/ArduPilot/ardupilot/pull/13207

Hwurzburg, 2020-06-25:
anyone that provides a doc on these just should add a DelyedMerge label on it....hope somone voluteers to work on the docs so they could actually be used by someone....

Hwurzburg, 2020-07-04:
@WickedShell Michael, can you review #2447 for accuracy and then close this if its covered everything here?