ArduPilot/ardupilot_wiki #2451: Using Gazebo Simulator with SITL - "make -j4" gives an error!

anish-pratheepkumar, 2020-01-15:
[Using Gazebo Simulator with SITL](https://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

**Hai,**
I was following the above-mentioned wiki. All worked well fine untill "make -j4" played spoil sport and ended up with the following error : -
*******************************************************************************
-- Build files have been written to: /home/anish/ardupilot_gazebo/build
anish@studienarbeit:~/ardupilot_gazebo/build$ make -j4
Scanning dependencies of target ArduPilotPlugin
Scanning dependencies of target ArduCopterIRLockPlugin
[ 25%] Building CXX object CMakeFiles/ArduCopterIRLockPlugin.dir/src/ArduCopterIRLockPlugin.cc.o
[ 50%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o
[ 75%] Linking CXX shared library libArduCopterIRLockPlugin.so
[ 75%] Built target ArduCopterIRLockPlugin
In file included from /usr/include/boost/variant.hpp:22:0,
                 from /usr/include/sdformat-6.2/sdf/Param.hh:26,
                 from /usr/include/sdformat-6.2/sdf/Element.hh:25,
                 from /usr/include/sdformat-6.2/sdf/sdf.hh:4,
                 from /home/anish/ardupilot_gazebo/src/ArduPilotPlugin.cc:41:
/usr/include/boost/variant/get.hpp: In instantiation of ‘typename boost::add_reference<T>::type boost::strict_get(boost::variant<T0, TN ...>&) [with U = short unsigned int; T0 = bool; TN = {char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long unsigned int, unsigned int, double, float, sdf::Time, ignition::math::v4::Color, ignition::math::v4::Vector2<int>, ignition::math::v4::Vector2<double>, ignition::math::v4::Vector3<double>, ignition::math::v4::Quaternion<double>, ignition::math::v4::Pose3<double>}; typename boost::add_reference<T>::type = short unsigned int&]’:
/usr/include/boost/variant/get.hpp:284:25:   required from ‘typename boost::add_reference<T>::type boost::get(boost::variant<T0, TN ...>&) [with U = short unsigned int; T0 = bool; TN = {char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long unsigned int, unsigned int, double, float, sdf::Time, ignition::math::v4::Color, ignition::math::v4::Vector2<int>, ignition::math::v4::Vector2<double>, ignition::math::v4::Vector3<double>, ignition::math::v4::Quaternion<double>, ignition::math::v4::Pose3<double>}; typename boost::add_reference<T>::type = short unsigned int&]’
/usr/include/sdformat-6.2/sdf/Param.hh:300:31:   required from ‘bool sdf::Param::Get(T&) const [with T = short unsigned int]’
/usr/include/sdformat-6.2/sdf/Element.hh:454:7:   required from ‘std::pair<T, bool> sdf::Element::Get(const string&, const T&) const [with T = short unsigned int; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/anish/ardupilot_gazebo/src/ArduPilotPlugin.cc:879:57:   required from here
/usr/include/boost/variant/get.hpp:212:5: error: static assertion failed: boost::variant does not contain specified type U, call to boost::get<U>(boost::variant<T...>&) will always throw boost::bad_get exception
     BOOST_STATIC_ASSERT_MSG(
     ^
CMakeFiles/ArduPilotPlugin.dir/build.make:62: recipe for target 'CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o' failed
make[2]: *** [CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/ArduPilotPlugin.dir/all' failed
make[1]: *** [CMakeFiles/ArduPilotPlugin.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
*****************************************************************************

Can anyone help me to figure out what went wrong !

Thanks in advance !

Mattis52, 2020-06-08:
I get similar errors :thinking:

Mattis52, 2020-06-09:
https://github.com/khancyr/ardupilot_gazebo/issues/11#issuecomment-609635517
This worked for me, but only on khancyr's build, not SwiftGust's

RawanElsayed00, 2024-05-20:
Hi I tried https://github.com/ethereum/solidity/issues/4124, changing <uint16_t> to <uint32_t> but I am getting this error:
 [ 75%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o
[ 75%] Built target ArduCopterIRLockPlugin
/home/rawanelsayed/ardupilot_gazebo/src/ArduPilotPlugin.cc: In member function ‘bool gazebo::ArduPilotPlugin::InitArduPilotSockets(sdf::v9::ElementPtr) const’:
/home/rawanelsayed/ardupilot_gazebo/src/ArduPilotPlugin.cc:978:40: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to a value of type ‘uint16_t’ {aka ‘short unsigned int’}
  978 |                         this->dataPtr->fdm_port_in, 9002);
      |                         ~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/rawanelsayed/ardupilot_gazebo/src/ArduPilotPlugin.cc:64:21: note:   initializing argument 3 of ‘bool getSdfParam(sdf::v9::ElementPtr, const string&, T&, const T&, const bool&) [with T = unsigned int; sdf::v9::ElementPtr = std::shared_ptr<sdf::v9::Element>; std::string = std::__cxx11::basic_string<char>]’
   64 |                  T &_param, const T &_defaultValue, const bool &_verbose = false)
      |                  ~~~^~~~~~
/home/rawanelsayed/ardupilot_gazebo/src/ArduPilotPlugin.cc:980:40: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to a value of type ‘uint16_t’ {aka ‘short unsigned int’}
  980 |                         this->dataPtr->fdm_port_out, 9003);
      |                         ~~~~~~~~~~~~~~~^~~~~~~~~~~~
/home/rawanelsayed/ardupilot_gazebo/src/ArduPilotPlugin.cc:64:21: note:   initializing argument 3 of ‘bool getSdfParam(sdf::v9::ElementPtr, const string&, T&, const T&, const bool&) [with T = unsigned int; sdf::v9::ElementPtr = std::shared_ptr<sdf::v9::Element>; std::string = std::__cxx11::basic_string<char>]’
   64 |                  T &_param, const T &_defaultValue, const bool &_verbose = false)
      |                  ~~~^~~~~~
make[2]: *** [CMakeFiles/ArduPilotPlugin.dir/build.make:76: CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/ArduPilotPlugin.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

Does anyone know how to fix it? Thanks in Advance

murphym18, 2024-05-23:
@RawanElsayed00 What version of gazebo are you using?

When I was trying to build this for gazebo 11 I ran into numerous issues including the one you linked to [4124](https://github.com/ethereum/solidity/issues/4124). But then I tired using gazebo 9 and I was able to build it, install it, and run it.