ArduPilot/ardupilot_wiki #2453: document Hott telem support

tridge, 2020-01-16:
we need to document the new Hott telem support. Ralf Helbing will provide some images for it

Ralfde, 2020-01-16:
Baudrate 19200 baud
Connect Graupner Telemetry connector of the receiver with an UART_TX. Best is to use an UART without RTS, CTS.
Here some screenshots of the mz-16/mz-32
[Screenshots.zip](https://github.com/ArduPilot/ardupilot_wiki/files/4072464/Screenshots.zip)

Ralfde, 2020-01-16:
Supported sensors protocolls from ardupilot are the sensors: GPS, EAM and Vario.
So additional an ESC and/or GAM telemetry sensor could be connected.

Hwurzburg, 2020-01-16:
A phot of the system would be good...maybe OmnibusF4 connected to RX sitting next to TX

Ralfde, 2020-01-16:
[image]
The foto does show an Omnibusf4v6.
SUMD or SBUS is connected to the LED connector at this fc.
Telemetry is connected to TX3 in the foto.
Laser rangefinder at RX4.
GPS at TX6, RX6 and I2C

Hwurzburg, 2020-01-17:
> Supported sensors protocolls from ardupilot are the sensors: GPS, EAM and Vario.
> So additional an ESC and/or GAM telemetry sensor could be connected.

sorry, I am afraid I am not familiar with those sensors in Ardupilot ...you will have to be more specific....as far as I know, AP has no support for an external VARIO  or Graupners EAM or GAM modules ....exactly how would such a sensor be connected? to the FC? to the HOTT telem signal line? I assume its a polled half duplex arrangement like FRSKY...
it would be nice to state what exactly what data is sent by the HOTT protocol from the autopilot and can be displayed on the TX

Ralfde, 2020-01-17:
Yes, SERIALn_OPTION=4 is halfdublex.
The GPS is the external sensor. The telemetry data of the GPS are send via HoTT telemetry to the HoTT transmitter and can be displayed there or optional in a smartphone app like the HoTT Viewer 2.
The barometer altitude datas are also send in the Vario protocoll together with the flight mode, disarmed, armed mode and Preflight check success, failed.
The battery datas like voltage, current and capacity is send as EAM data to the radio, if such a sensor is included in the flight control or connected externally to the fc.
I think the screen shots do show most of the datas, which are supplied.

Hwurzburg, 2020-01-17:
okay...what you said implied that other sensors besides the normal ones in Ardupilot could be added....what you meant was that their data was formatted as HOTT sensors and transmitted...okay, so nothing needs be added to the PR I am about to push....thanks

Hwurzburg, 2020-01-17:
#2461 awaiting review

Ralfde, 2020-01-18:
Correct. The HoTT Sensor bus allow to connect different sensors parallel.

closed 2020-01-18 by Hwurzburg