ArduPilot/ardupilot_wiki #2454: Guided no gps mode - how to use it

anish-pratheepkumar, 2020-01-16:
[Copter Commands in Guided Mode](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Good Day,
I read "SET_ATTITUDE_TARGET -  this is the only message accepted by Guided_NoGPS" and the body pitch, roll and yaw rate inputs are disabled. which mean I cannot control that from Guided_NoGPS mode. The only way is to give it X, Y,Z position target and it will move to that position. Is this correct?

If it is correct it means I need an **initial position lock** and then a gps denied position estimation device like Optical flow(which I have). So does it mean I need a GPS to get initial position lock?

Thanks in advance!

Hwurzburg, 2020-01-16:
sorry this is a support question, not a wiki issue....please ask in the discuss.ardupilot.org forums...sorry but must close this...

closed 2020-01-16 by Hwurzburg

rmackay9, 2020-01-16:
@anish-pratheepkumar, as Henry says these questions should go into the Copter-4.0 support forums: https://discuss.ardupilot.org/c/arducopter/copter-40 but in any case, you'll need to set the EKF origin which can be done by opening Mission Planner's Flight Data screen, then right-mouse-button-click on the map and select, "Set Home Here >> Set EKF Origin Here"

anish-pratheepkumar, 2020-01-16:
> @anish-pratheepkumar, as Henry says these questions should go into the Copter-4.0 support forums: https://discuss.ardupilot.org/c/arducopter/copter-40 but in any case, you'll need to set the EKF [...]
Hey, Thank you for your response!
I have tried that a lot of times and "ppoirier" also mentioned the same in ardupilot discussions. Unfortunately, it doesn't work for me at all! I wonder if maybe because I don't have a GPS connected in my setup. Have any of you guys tried this "Set Home Here >> Set EKF Origin Here" without GPS connected to your copter?

 I also don't understand the technical concept as in how do we get a position lock if we don't have a GPS connected to our system, is it like giving a fake origin?

Any help will be great, I have been spending months now on this, still no success. If I could just figure out what is the reason for this not working!

rmackay9, 2020-01-17:
@anish-pratheepkumar,

no, it shouldn't require a GPS.  Just an optical flow and a downward facing lidar should be enough.

I did this just yesterday on a Rover and it worked.

I would not confuse things by trying to involve mavros at this stage. Just stick with Mission Planner because we know that works out-of-the-box (ROS has all kinds of setup that could break things).

I'm afraid we really need to move this to the support forums so I won't reply anymore here.  The best place is the gitter forum (https://gitter.im/ArduPilot/VisionProjects) or the Copter-4.0 forum: https://discuss.ardupilot.org/c/arducopter/copter-40