ArduPilot/ardupilot_wiki #2474: Clarify OneShot/DShot testing procedure

rmackay9, 2020-01-24:
This issue emerged as part of this discussion: https://discuss.ardupilot.org/t/oneshot125-42-blhelis/51462

We should document a testing procedure so that users of OneShot and DShot can confirm that the outputs from the flight controller are really working using the desired protocol and not sending the default PWM signals in the 1000 to 2000 range.

The confusion comes because many ESCs will fall back to accepting the regular default PWM range (about 1000 to 2000) if they don't receive the OneShot or DShot protocol so simply testing that the motors/propellers spin is not sufficient.  Also it is not clear if the **chXout** values visible in MP's status screen (i.e. the values from the [SERVER_OUTPUT_RAW](https://mavlink.io/en/messages/common.html#SERVO_OUTPUT_RAW) messages) and onboard log RCOU messages change when OneShot or DShot are used or if they are always in the 1000 to 2000 range

SergeyBokhantsev, 2020-01-24:
Could you please also clarify for DShot protocol: does ArduPilot uses full 2048 steps resolution?
My concern is that core code may produce values in a 1000-2000 range. And then for DShot the value just scaled to 0-2048 range. Thus smallest step is 2 unit.

rmackay9, 2020-01-25:
Here is another support issue on 4.0 to do with users unsure if DShot is being used or not: https://discuss.ardupilot.org/t/pixhawk-4-holybro-arducopter-3-6-9-4-0-1rc3-no-control-over-two-motors-in-an-x8-with-dshot/51480

rmackay9, 2020-01-28:
@SergeyBokhantsev, yes, I suspect that is the case, @tridge would know for sure.

Hwurzburg, 2020-04-14:
resolved with GCS message that has been added

closed 2020-04-14 by Hwurzburg