ArduPilot/ardupilot_wiki #2477: Low IMU publishing rate when using Ardupilot SITL with ROS

erinline, 2020-01-27:
[Data from AP to ROS](https://ardupilot.org/dev/docs/ros-data-from-ap.html)

When running Ardupilot in SITL with ROS, mavros IMU topic publishes at a rate of about 4Hz, even when I manually reset it to higher rates, it resets again to 4Hz. The above link suggests that the reason for this has to do with APWeb, which I was not even aware I was using. The instructions on that page for how to resolve it are minimal, if anyone has successfully resolved this issue, how did you do so?

BTW, I have looked at my computer's /etc/rc.local file and it says nothing about APWeb (more to the point, it says nothing significant about anything). Other suggestions?

khancyr, 2020-01-27:
As stated in the comment, you only have APWeb if you are using APSync which shouldn't the case for SITL.
I am pretty sure that you launch SITL with sim_vehcile.py that, by default launch mavproxy. So, you need to say to mavproxy to not reset the streamrate or connect directly to SITL with tcp, in that case, you will have full control on the streams as mavproxy won't be between you and SITL

khancyr, 2020-01-27:
I am letting this open as this part of documentation could be improved a little bit.

erinline, 2020-01-27:
Thanks for the info @khancyr ! So how would I make mavproxy not reset the streamrate?

erinline, 2020-01-28:
Update: I have figured out how to set streamrate so that it does not get overridden, but this applies to all data streams. In any case, it's an improvement. For any others looking for the same: start ArduPilot SITL with `../Tools/autotest/sim_vehicle.py --mavproxy-args="--streamrate=50"`, ofc replacing 50 as desired. It would be ideal to just change the IMU data stream rate though. Still trying to figure that one out.

khancyr, 2020-01-28:
Hello, 
Just set the streamrate to -1 on mavproxy to disable rate auto setting and do the management either with mavros or parameters