ArduPilot/ardupilot_wiki #2480: UAVCAN ESC page should explain feedback from ESC can be found in CESC log message

rmackay9, 2020-01-28:
This comes from [this user request asking how to view feedback from CANESCs](https://discuss.ardupilot.org/t/in-what-section-in-log-can-i-find-uavcan-escs-telemetry/51613).

We should add a note (and perhaps a picture) telling the user to look at the CESC onboard log message for post-flight analysis or the ESC_TELEMETRY message for inflight viewing of the output.  I don't actually know if MP displays the results from the ESC_TELEMETRY message anywhere but if it does it would appear in the Flight Data screen's Status tab.  worst case the user could look at the Mavlink Inspector.

Hwurzburg, 2020-01-28:
does this telem/log message apply also to both CAN ESCs and UAVCAN ESCs?

rmackay9, 2020-01-28:
@Hwurzburg, yes, I think that all CAN ESCs (UAVCAN, KDECAN, ToshibaCAN) will send the RPM (reported by the ESC) to the ground station using the [ESC_TELEMETRY_xxx_TO_yyy](https://mavlink.io/en/messages/ardupilotmega.html#ESC_TELEMETRY_1_TO_4) mavlink message.

As for onboard logging it's a less happy story.  It seems that UAVCAN ESCs are logging using the CESC message ([see Write_ESCStatus here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Logger/LogFile.cpp#L809)) while BLHeli, KDECAN, PiccoloCAN and ToshibaCAN write using the "ESC" onboard log message ([see Write_ESC here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Logger/LogFile.cpp#L770))

rmackay9, 2020-01-28:
I've pushed a couple of commits to resolve this I think, thanks!

closed 2020-01-28 by rmackay9