ArduPilot/ardupilot_wiki #2495: SBUS Servos page should mention BRD_SBUS_OUT parameter

rmackay9, 2020-01-30:
It seems that the SBus Servos page neglects to mention that for the "SB" or "SBus" output to work on Pixhawk/Cube autopilots the BRD_SBUS_OUT parameter must be set to a non-zero value.  https://ardupilot.org/copter/docs/common-sbus-out.html

I have never actually used the SBUS out feature but hopefully @tridge can confirm this parameter must be set.  It certainly seems so from the parameter descriptions.

auturgy, 2020-01-30:
You also need to make sure RSSI isn‘t  enabled on that pin (103) from memory. I don’t think there’s a check.

Regards,

James

rmackay9, 2020-01-30:
I've just successfully tested the SBUS output using a method recommended by Tridge.  What I did was connect a 3-wire servo cable from the CubeBlack's SBUSo port to the RCIN port and then I could see on MP's Flight Data screen's status and tuning areas that the inputs were changing when I ran the motor test.  Some pics below show what I did.

[image]
[image]
[image]

I didn't bump into the RSSI issue mentioned by @auturgy but I did find that I needed to ensure the vehicle was armable.  In the end I set the following parameters:

- BRD_SBUS_OUT = 1 (50hz)
- FRAME_CLASS and FRAME_TYPE appropriate for my vehicle
- BRD_SAFETYENABLE = 0 (my vehicle doesn't have a safety switch)
- ARMING_CHECK = 0 (this was only necessary because I hadn't done the RC input calibration)

Hwurzburg, 2020-01-30:
#2498

closed 2020-01-30 by Hwurzburg

rmackay9, 2020-01-30:
awesome, thanks very much!

mattdholmes00, 2021-05-11:
silly question, but I'm not seeing BRD_SBUS_OUT in my parameters, what do I have to do to even see it so I can enable it?

rmackay9, 2021-05-11:
@mattdholmes00,

I suspect that the board you've got doesn't have an SBus out port but it is hard to be sure without more information (like an onboard log).  Really though, support should go to the forums. So for example, if you're using Copter-4.0.x then a post in this forum would be best: https://discuss.ardupilot.org/c/arducopter/copter-40/155