ArduPilot/ardupilot_wiki #2500: Add warning to RadioLink MiniPix page

rmackay9, 2020-01-31:
A few changes we should probably make to our RadioLink MiniPix wiki page. https://ardupilot.org/copter/docs/common-radiolink-minipix.html

- shrink the top image to no more than 450pixels wide
- add a warning that this board is not recommended because the manufacturer is not abiding by the GPLv3 license which requires making the modified ArduPilot software available to their customers.  Also the standard ArduPilot firmware will not run on these boards.

rmackay9, 2020-01-31:
@tridge has had the most contact with the RadioLink people so he may know better.  Still, I think we should actively steer our users away from this board because we just end up with support calls when users find they can't run our latest firmware on the boards.

Hwurzburg, 2020-01-31:
If the consensus is that we steer them away, we should reduce the page to just a warning that the board is incompatible and leave it at that

peterbarker, 2020-01-31:
@rmackay9 Why won't the standard ArduPilot firmware run on these boards?!  Rather news to me as I have several.

I don't mind the GPLv3 warning.

rmackay9, 2020-01-31:
For reference here's the open source notice on the Radio link website: http://radiolink.com.cn/doce/mini%20pix-open%20source.html

peterbarker, 2020-01-31:
OK, a little bit of a play here shows that RadioLink have released a "v1.0 II".

This does *not* work out of the box with ArduPilot master or ArduCopter 3.6.

What a pain.

peterbarker, 2020-02-09:
So.

Version 2 of the v1.0(!) has done away with the ramtron driver.  If you disable the ramtron and use hal-storage-empty, and change the reserved-flash-space to 32kB then the device boots normally.

Current theory is that RadioLink are using the second 16kB page for storage.

That's a problem in that you could lose all your parameters etc if you are unlucky.

We may end up using a patch tridge has lying around to use a second 16kB page to do things properly, or back the parameters up to SD card or something...

rmackay9, 2020-02-26:
Here's the warning that I think we should add to the page.  It's possible that it's a bit too harsh but I think it's accurate.

This autopilot is not recommended because the board is not compatible with the official ArduPilot software despite multiple efforts to work with the manufacturer to make it compatible.  The manufacturer is also apparently not abiding by the GPLv3 license which requires releasing the modified source code to its customers.

Hwurzburg, 2020-02-26:
#2581 addresses...closing

closed 2020-02-26 by Hwurzburg