ArduPilot/ardupilot_wiki #2503: Mention BLHeli ESC telemetry can be used as battery monitor

rmackay9, 2020-02-04:
This comes out of this discussion in the Copter-4.0 category: https://discuss.ardupilot.org/t/possible-5v-power-issues-kakute-f7-w-tekko32-4in1-ac4-0-1/51872/12

If BATT_MONITOR = 9 then BLHeli ESC Telemetry can be used as a source of voltage (and current?) for failsafes, etc.  We should add a note on the [DShot and BLHeli setup page](https://ardupilot.org/copter/docs/common-dshot.html) I think.

Hwurzburg, 2020-02-04:
As far as I know BLH telem only reports rpm and current, not voltage...and the current is only the motors' not the rest of the system...I dont think we should recommend to use it for failsafes...however, it can be displayed IF the autopilot has an integrated OSD and we do mention that...
I am not sure how the BATT_MONITOR function works with BLH ESC...does it report the AVERAGE current from all attached ESCs? or just the first one it polls? is it worth recommending as a system battery monitor if the information is partial or not the total?

auturgy, 2020-02-04:
It probably needs it’s own setup page/paragraph. Basically you need to use the SumBattery type, which adds current and averages voltage from other monitors, then add the escs underneath it.
What data the esc sends is dependant on the type of esc, but i th  the maximal set is voltage, instantaneous current, total consumed current, temperature and rpm.

Regards,

James

Hwurzburg, 2020-02-04:
-do you need a batt monitor instance per each ESC on the single telemetry line?or do they have to individually consume a separate RX UART input?
-the batt monitor only deals with the voltage and currents reported, not the temp or RPM, correct?
-the voltage capability has only been recently added, how does the batt monitor deal with only current and RPM being reported from the earlier ESCs?

rmackay9, 2020-02-06:
@Hwurzburg,

From a look at the driver just one battery monitor will monitor all the BLHeli ESCs.  So no need for example to setup 4 separate batt monitors for a quad.  Just setting BATT_MONITOR = 9 should do it.

I suspect that for small quads in particular the vast majority of the current is drawn by the motors so I don't think that users should need to setup both the normal analog and the BLHeli monitor.  Just the BLHeli monitor should be good enough.  The voltage (which is the most important) should be the same whether an analog or BLHeli ESC is measuring it so again I don't see the need for another battery monitor.  Someone who actually flies these small copters would be a better source of info of course but I suspect I've got this right.

Hwurzburg, 2020-02-12:
checked with Tridge...id you set a monitor for BLHeli it will average all connected ESCs for v, total current, and  accumulate total mah...will add to the posted  esc PR

Hwurzburg, 2020-02-13:
added to PR #2502

closed 2020-02-13 by Hwurzburg