ArduPilot/ardupilot_wiki #2538: The flight mode (Stabilize/AltHold) is being changed by some reason

jaguraphg, 2020-02-15:
Hello,

Could anybody guys help me to figure it out please? I try to configure the AltHold mode in Arducopter (this is my 2nd day with Ardupilot, was flying with INAV before.

The problem is that by some reason when I take off I need to pull the throttle more and more to climb, but it is not climbing. After moving the stick more the copter does very very sharp move up (about 5-6 meters for 0.1 sec), I have to release the throttle to 'catch' the copter in the air, and it seems hovering properly when the throttle is centered. But if I move the throttle up again then the copter hovers again until I move the stick more and more and then the copter flying up sharlply again.

I tried to analyze the logs and as I see the flight modes are being changed in flight between Althold and Stabilize. It seems this is the problem. Please see the logs.

https://drive.google.com/file/d/1MmyWCue_6grhoVfdkf5KHgz9Xo6_GVrz/view?usp=sharing

FC: Holybro Kakute F7.

I really do not have an idea what might be the reason. Any thoughts?

Thanks.

Hwurzburg, 2020-02-15:
fyi, discuss.ardupilot.org is the place for support questions....this is for issues with the wiki documentation...

second, you have your flight mode channel set to your throttle channel instead of a TX switch channel....you should carefully read thru the copter wiki sections of First Time Setup https://ardupilot.org/copter/docs/initial-setup.html and First Flight https://ardupilot.org/copter/docs/flying-arducopter.html

closing

closed 2020-02-15 by Hwurzburg

jaguraphg, 2020-02-15:
Thank you for reply. I configured the flight mode channel correctly I believe, but ok, I will ask the same question in the discussion board. Thanks.

Hwurzburg, 2020-02-15:
sorry its not....from your log:
FLTMODE1	0
FLTMODE2	0
FLTMODE3	0
FLTMODE4	2
FLTMODE5	0
FLTMODE6	5
FLTMODE_CH	3  
RCMAP_THROTTLE	3

its on throttle

jaguraphg, 2020-02-15:
Oh god, so you're right. How could it be done? I definitelly not changed something like that. And why I was sure that the channel is correct because I see the mode changed in the mission planner. But it seems that the mission planner does not take this setting into account. Then it answers to all quesstions.

I will check this in the morning, please excuse me for writing here. Thank you very much again!