ArduPilot/ardupilot_wiki #2551: Duplicate Node com.hex.here../125!

coptrdr2, 2020-02-19:
[Search](https://ardupilot.org/copter/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Hwurzburg, 2020-02-19:
without detailed information, this will be closed tomorrow

coptrdr2, 2020-02-19:
When attempting to fly a quadcopter with  Arducopter 4.02 firmware and CAN GPS firmware 1.3.19 we keep getting duplicate node com.hex.here../125! prearm error.  The GPS is configured with the switch in the CAN position and the pixhawk is set to GPS_Type to 9 UVCAN.. We have built several (75+) machines without any problems.

Hwurzburg, 2020-02-19:
Sorry you are having an issue...setup and support issues and questions should be entered in the appropriate support forum: discuss.ardupilot.org...
this area (GitHub) is for corrections or additions to the Wiki documentation...
closing

closed 2020-02-19 by Hwurzburg