ArduPilot/ardupilot_wiki #2558: Automatic rudder control

uavengine, 2020-02-21:
[Flight Features](https://ardupilot.org/plane/docs/flight-features.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

khancyr, 2020-02-21:
hello,

Without question or pointing what is wrong, we will close this issue

uavengine, 2020-02-22:
Thank you for the notification:

I have a question Mission planner(It contains so many parameters) about the parameter for Takeoff mode.
I was not clearabout the servomotor control.

In manual mode,AILRON and Rudder move independently,

But if we setTakeoff mode, FBWA or Takeoff mode, Auto,

If we move lever (Ail)of Prop, then both AILRON and Rudder move simultaneously that motioncaused instability during Taxiing.

If it moved AILRONand Elevator upward, it will be stable.   How we can set thecombination of servo

(AIL,RUDDER)-->(AIL, Elev upward)

 

This is myquestion

Hwurzburg, 2020-02-22:
@uavengine Sorry but this is for issues with the Wiki documentation, not support questions. Please ask those in the discuss.ardupilot.org forums for assistance with your application. However, in non-Manual modes, the parameter KFF_RDDRMIX is used to mix rudder with aileron input.

closed 2020-02-22 by Hwurzburg