ArduPilot/ardupilot_wiki #2633: Setting up the waf Build Environment on Windows using Cygwin

meysamuav, 2020-03-23:
hello,
i want to start "stil" on pixhawk cube.
I did every tip for install Cygwin. all done but for activate and install waf, the tutorial isn't enough and clear, please give me a tutorial with more details.
thanks
[image]

[Setting up the waf Build Environment on Windows using Cygwin](https://ardupilot.org/dev/docs/building-setup-windows-cygwin.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

meysamuav, 2020-03-23:
[image]

stephendade, 2020-04-19:
Cywin instructions updated in https://github.com/ArduPilot/ardupilot_wiki/pull/2683