ArduPilot/ardupilot_wiki #2641: Setting up the Build Environment (Linux/Ubuntu) needs updating

TheSoundMan000, 2020-03-26:
Referring to the comments made by Eosbandi way back in '18 at https://discuss.ardupilot.org/t/build-fails-with-wsl-on-windows-10/31555. Relevant extracts with edits as per his following comments provided.

> emove arm-none-earbi-ar from the path in ~/.profile
> apt-get install gcc-arm-none-eabi
> reload .profile

This step is still necessary for an installation on Ubuntu 18.0.4 x64, the installer seems to install the wrong version or installs it incorrectly and continually throws missing package error. Better yet would of course be a fix to the installation script.

khancyr, 2020-03-26:
Hello,

I will look back at our instructions but using the Ubuntu toolchain is wrong

Hwurzburg, 2020-03-26:
not my area of expertise for sure, but I am not sure why its necessary for an Unbuntu install under WSL and not for pure Linux...I will test today on my fresh WSL install on Win10

TheSoundMan000, 2020-03-26:
Not using wsl, the solution on that page just happens to be relevant. The instructions on the wiki page relate to installation on native Ubuntu.

khancyr, 2020-03-26:
`apt-get install gcc-arm-none-eabi` is wrong that is an untested toolchain and is regularly broken. 
The wiki instructions are correct.

Hwurzburg, 2020-03-26:
I just installed the toolchain in an Ubuntu under WSL using the instructions in the Wiki : https://ardupilot.org/dev/docs/building-setup-linux.html

works fine, including SITL and MaxProxy....wiki instructions are correct
closing this

khancyr, 2020-03-26:
@Hwurzburg, the test would have been to build for a CubeBlack or another chibios target.

 Anyways, there wasn't any evidence of error and the proposed fix is wrong.

TheSoundMan000, 2020-03-26:
Just did a **another** fresh install of Ubuntu 18.0.4 Desktop and followed the instructions outlined in the wiki: https://ardupilot.org/dev/docs/building-setup-linux.html. This has now been tested on a virtual machine and 2 seperate PCs: same failure, same error. the error most definitely exists. It is not possible to build for a pixhawk 4 (or cube or I suspect any Chibios target as @khancyr says above) due to the missing tool chain.

The proposed fix may (is probably) be wrong, I'm not familiar enough with ardupilot to know however it is a solution that is currently working for myself. As noted it is untested and not at all an ideal solution however given the current instructions do not allow for a successful build its my only bet. If someone familiar with the requirements could amend the instructions with appropriate corrections it would be greatly appreciated.

TheSoundMan000, 2020-03-26:
> sudo apt-get update
> sudo apt-get install git
> sudo apt-get install gitk git-gui
> git clone https://github.com/ArduPilot/ardupilot
> cd ardupilot
> git submodule update --init --recursive
> sudo Tools/environment_install/install-prereqs-ubuntu.sh -y
> . ~/.profile

Did a restart here.

> cd ardupilot
> sh
> ./waf configure --board CubeBlack (or pixhawk4 etc)

Fails due to missing toolchain. Fix implemented using untested toolchain:

Remove arm-none-earbi-ar from the path in ~/.profile

> apt-get install gcc-arm-none-eabi

reload .profile

> ./waf configure --board CubeBlack
> ./waf copter

Now everything exectutes currectly.

TheSoundMan000, 2020-03-26:
I've just re-read the comments by @Hwurzburg and I suspect the intent of my issue was not understood by your return to the WSL matter. Just to re-iterate my earlier comments and hopefully progress the issue here: I am not attempting to install on a windows computer. I am referring to a completely native Ubuntu installation (ie ubuntu OS, **not WSL**).

Hwurzburg, 2020-03-27:
I just tested on a new native linux only box...no issues there either following the documentation in the wiki https://ardupilot.org/dev/docs/building-setup-linux.html

TheSoundMan000, 2020-03-27:
@Hwurzburg can you confirm your Ubuntu version and that the installation process was as described above (or the material deviations if there are any so I am able to give that a shot my end). Could you also provide the expected location of the toolbox installation in the standard installation and the expected PATH changes so I can cross check both of these?

If you could re-open the issue so that others may contribute would also be appreciated, I'd like to get this resolved if possible.

Hwurzburg, 2020-03-27:
Ubuntu 18.04LTS, instal exactly as per the link...PATH variable
     1	/home/henry/Desktop/ardupilot/Tools/autotest
     2	/opt/gcc-arm-none-eabi-6-2017-q2-update/bin
     3	/home/henry/.local/bin
     4	/usr/local/sbin
     5	/usr/local/bin
     6	/usr/sbin
     7	/usr/bin
     8	/sbin
     9	/bin
    10/usr/games
    11/usr/local/games
    12/snap/bin
    13/home/henry/.local/bin
    14/home/henry/.local/bin

TheSoundMan000, 2020-03-27:
Well blowed if I know whats going on, I've just installed WSL and followed the same process and everything's rosy on the stock toolchain whilst the straight Ubuntu box fails. I'm out of ideas.

khancyr, 2020-03-27:
Hello,

Why do you use sh ?

Hwurzburg, 2020-03-27:
I just noticed that....doesnt that create a new environment of some kind? maybe thats whats different...we dont do that command

TheSoundMan000, 2020-03-27:
You know you might have just solved it because that's something I've not done in WSL, testing now. I'm not sure where I've dragged it up from I'll be honest, I thought it was in the wiki but I've just checked and it definitely isn't so I'm not sure. Will update.

khancyr, 2020-03-27:
using sh will use sh as shell and not bash ... so the .profile won't be loaded and thus the path for our toolchain is unknown. 
So the instructions are correct ! Please don't use the arm-none-eabi provided by Ubuntu as it doesn't work (it will compile but you will have a runtime error due to some issue on linker ...)

closed 2020-03-27 by khancyr

TheSoundMan000, 2020-03-27:
Instructions are correct and it works, thanks for all your help guys! I can't work out where I got the sh thing from but oh well. Glad to see its all working.