ArduPilot/ardupilot_wiki #2644: Issues

rgtoth, 2020-03-26:
It would be great if there was a dedicated page for acronyms used in each manual and and words or info they represent. I know many but there is always a few I dont know.

Thanks

Hwurzburg, 2020-03-26:
an acronym should be spelled out before later use....eg "Mission Planner (MP) is a common ground control station. You will need to install MP before using it."
Please let us know when you run across a document that does not do that, and we will resolve...trying to compile a list of acronyms would no only be daunting,and probably incomplete, but almost immediately out of date as new features get added...its better to fix the violations to the above policy when noted, if possible...feel free to add an issue any time you find one...thanks

Hwurzburg, 2020-07-06:
no more inputs on this so...closing...feel free to point out places where an acryonm is used without spelling it out first...

closed 2020-07-06 by Hwurzburg