ArduPilot/ardupilot_wiki #2654: Description of tailsitter gain scaling options is wrong

kd0aij, 2020-04-03:
This has apparently been out of date since last April:
https://github.com/ArduPilot/ardupilot/commit/95da094bfec00a36f46b9fc3ba0c0996d1d619ad

I thought I had updated this though, with a table of entries for each value of the gainscaling mask...
Will see if I can find the wiki commit

Hwurzburg, 2020-04-04:
Mark, first we need to make sure its correct for stable..let me know what we need to change....then we can put a pr in with delay merge for current master behavior....unless we can specify the exact future stable release number, then we can add with a note for the changed behavior in future Plan 4.x release...

kd0aij, 2020-04-04:
I think it's correct for ArduPlane-4.0.5
Will double check and generate the updated description for master.   I think putting it in as a note is a good idea for so people flying master don't get confused like me :)