ArduPilot/ardupilot_wiki #2680: Add warning to AutoTune page

rmackay9, 2020-04-16:
We fairly often get user complaints that AutoTune produced a bad tune that led to a crash.  I think we should replace the "Position Hold during AutoTune is available in Copter 3.5 (and higher)." at the top with:

Warning: AutoTune is not always able to determine a good tune for the vehicle and may result in gains that lead to an unflyable vehicle that may crash.

I'm happy to adjust the wording but I think we need to be honest with our users that even if they follow the instructions, they may end up with an vehicle that flies worse than it did when they started.

FYI @lthall

rmackay9, 2020-04-16:
Here's some more improvements we should make to the AutoTune and Tuning Process Instructions guides based on a discussion with Leonard:

- add links between the two pages when they talk about each other (i.e. if "AutoTune" appears on the Tuning Process Instruction page it should link to the AutoTune page
- the AutoTune page needs a section on testing post autotune.  This should include a link to the [Tuning Process page's Evaluating the Aircraft Tune](https://ardupilot.org/copter/docs/tuning-process-instructions.html#evaluating-the-aircraft-tune) section from the AutoTune page.  This should include the important advice that the first takeoff after an AutoTune should be in AltHold or Stabilize mode.
- the [Tuning Process page has a section on AutoTune](https://ardupilot.org/copter/docs/tuning-process-instructions.html#autotune) that has information and warnings that are strangely not on the AutoTune page itself.  We should move, duplicate and/or link these together.
- the tuning processes's section, "Some signs that Autotune has been successful are:" should probably be reversed so that it's "Signs that the AutoTune has failed".

rmackay9, 2020-05-18:
I've added the warning at the top of the AutoTune page to lead users towards the Tuning Process Instructions.  Hopefully we can get the other changes made at some point in the near-ish future.

Hwurzburg, 2020-05-18:
@rmackay9 re-read the section:  "Some signs that Autotune has been successful are:" 
I believe that the title is correct...after a tune Ps will have been increased, and D is above min tune D

the next section on excessive responsivenes is confusing though...should the Ps be lowered if above 10, down to 6? or should they be lowered to the 10-6 range if above 10? etc.

Hwurzburg, 2020-05-18:
Updated #2760 with information you requested be added....

closed 2020-06-21 by Hwurzburg