ArduPilot/ardupilot_wiki #2689: Improvements to Motor Ordering Page

rishabsingh3003, 2020-04-23:
The following page: https://ardupilot.org/copter/docs/connect-escs-and-motors.html#connect-escs-and-motors contains the details on how to connect ESC to Motors with the relevant ordering. But I feel the images and the details are very limited here. The [FRAME_TYPE](https://ardupilot.org/copter/docs/parameters.html#frame-type-frame-type-x-v-etc) parameter has a lot of other useful ordering, such as ClockwiseX that might interest a user.

I feel we should discuss the FRAME_TYPE and FRAME_CLASS parameters to some length here.

rmackay9, 2020-05-02:
Thanks for raising this.  We've actually already got an issue to cover this I think so I'll close this one. https://github.com/ArduPilot/ardupilot_wiki/issues/1875

closed 2020-05-02 by rmackay9

rishabsingh3003, 2020-05-02:
@rmackay9 thanks for pointing that out! I did not know it already existed.