ArduPilot/ardupilot_wiki #2700: missing description of ch6 TUNE knob usage

tridge, 2020-04-30:
We don't seem to have a page on tuning params with CH6 TUNE parameter for copter. There is some info for heli, but not copter as far as I can see

rmackay9, 2020-04-30:
This is all we have and it's more about using the Ch6 tuning knob for roll and pitch rate tuning. https://ardupilot.org/copter/docs/ac_rollpitchtuning.html#in-flight-tuning

Hwurzburg, 2020-04-30:
If I can get a quick list of params associated with each below, I can create a transmitter tuning section like in plane....no formatting needed, just a simple list:

0 | None
-- | --
1 | Stab Roll/Pitch kP
4 | Rate Roll/Pitch kP
5 | Rate Roll/Pitch kI
21 | Rate Roll/Pitch kD
3 | Stab Yaw kP
6 | Rate Yaw kP
26 | Rate Yaw kD
56 | Rate Yaw Filter
55 | Motor Yaw Headroom
14 | AltHold kP
7 | Throttle Rate kP
34 | Throttle Accel kP
35 | Throttle Accel kI
36 | Throttle Accel kD
12 | Loiter Pos kP
22 | Velocity XY kP
28 | Velocity XY kI
10 | WP Speed
25 | Acro RollPitch kP
40 | Acro Yaw kP
45 | RC Feel
13 | Heli Ext Gyro
38 | Declination
39 | Circle Rate
41 | RangeFinder Gain
46 | Rate Pitch kP
47 | Rate Pitch kI
48 | Rate Pitch kD
49 | Rate Roll kP
50 | Rate Roll kI
51 | Rate Roll kD
52 | Rate Pitch FF
53 | Rate Roll FF
54 | Rate Yaw FF
57 | Winch
58 | SysID Magnitude

rmackay9, 2020-05-02:
@Hwurzburg, I think I've filled in all the corresponding parameter names.  Apologies in advance if I got some wrong!

Hwurzburg, 2020-05-02:
thanks! super!

Hwurzburg, 2020-05-03:
very close...thanks...
#2711

closed 2020-05-12 by Hwurzburg