ArduPilot/ardupilot_wiki #2708: Ekf Failsafe Problem for SITL

staycoolish, 2020-05-02:
Hello, I am following the basic tutorial: https://ardupilot.org/dev/docs/rover-sitlmavproxy-tutorial.html#

After writing **arm throttle**, I try to set a position with right click, and say **fly to** , but keep getting the failsafe error. What is the reason?
[image]

Thanks!

Hwurzburg, 2020-05-02:
I suggest you ask in the support forum: discuss.ardupilot.org...this is for documentation errors and updates..better change to get a response over there..thanks...
will close this

closed 2020-05-02 by Hwurzburg

rmackay9, 2020-05-04:
@staycoolish, I don't think the failsafe error is related actually. It seems more likely that there's an issue with how the fly-to command is being sent.  It might help to update mavproxy.. i'm not sure, it's just a guess.

rmackay9, 2020-05-04:
@staycoolish, the issue is probably that the vehicle is not in Guided mode.  AP generally only accepts fly-to commands while in guided mode.

staycoolish, 2020-05-04:
@rmackay9 Thanks a lot! You are right, I forgot to write GUIDED, although it was right in front of me :D good luck with other development issues!

rmackay9, 2020-05-04:
@staycoolish, I suspect the wiki led you astray.  I corrected the wiki page after you reported this issue so hopefully the next person won't trip over this.  thanks for the report!

staycoolish, 2020-05-05:
@rmackay9 That makes better sense, I'm happy to helpful! By the way, there is another small issue in that wikipage, in the current ardupilot, there is no folder named **APMRover** , but there is a folder named **Rover** , which is the only Rover-related folder. In the wiki page, this is written:
 
```
cd ~/ardupilot/APMrover2 
sim_vehicle.py --map --console
```

However, it should be this:

staycoolish, 2020-05-05:
@rmackay9 That makes better sense, I'm happy to be helpful! By the way, there is another small issue in that wikipage: In the current ardupilot, there is no folder named **APMRover2** , but there is a folder named **Rover** , which is the only Rover-related folder. In the wiki page, this is written:

```
cd ~/ardupilot/APMrover2 
sim_vehicle.py --map --console
```

However, it should be this ( since there is no APMRover2):

```
cd ~/ardupilot/Rover 
sim_vehicle.py --map --console
```

Maybe writing **GUIDED** became necessary, due to the switching from APMRover2 to Rover.

Hwurzburg, 2020-05-05:
The recent name change to Rover for the folders has not been update in the developers section of the wiki yet...its on my todo list....

peterbarker, 2020-05-05:
On Tue, 5 May 2020, staycoolish wrote:

> Maybe writing GUIDED became necessary, due to the switching from APMRover2 to Rover.

Nope, that's not going to be the case.

PR to update the documentation: 
https://github.com/ArduPilot/ardupilot_wiki/pull/2716

Peter

staycoolish, 2020-05-05:
@peterbarker I am not very familiar with github, linux, and ros; but trying to catch up. So I searched how to Pull Request, then made a fork of ardupilot_wiki, and cloned it. Then I saw that you already made dozens of corrections; and since they aren't visible on the webpage, I assume someone must approve them, or maybe you refresh the webpage at every x days. Anyway, I will try PR next time, but thanks for the advice, it doesn't seem too hard really.

peterbarker, 2020-05-05:
On Tue, 5 May 2020, staycoolish wrote:

> @peterbarker I am not very familiar with github, linux, and ros; but trying to catch up. So I searched how to Pull Request, then made a fork of

Those are all awesome technologies - well worth adding them to your 
toolkit.

> ardupilot_wiki, and cloned it. Then I saw that you already made dozens of corrections; and since they aren't visible on the webpage, I assume someone must
> approve them, or maybe you refresh the [...]

Yes, we use a simple Pull Request workflow using github; @hwurzburg is our 
Wiki maintainer and will hopefully check through the changes, see that 
they're OK (and forgive me for a few under-the-radar modifications...) and 
merge the PR.

Thanks for thinking of doing a PR - I'm a little sorry I got in first on 
this one.  If you do wish to get started on some simpler things we've 
found lacking in ArduPilot, you might check our "Good First Issues" list 
of issues: 
https://github.com/ArduPilot/ardupilot/issues?q=is%3Aopen+is%3Aissue+label%3A%22good+first+issue%22

There are also a whole bunch of issues against the Wiki if you were 
specifically interested in that aspect of the project.

Yours,

Peter

staycoolish, 2020-05-05:
Thanks for the detailed explanation and your humility @peterbarker ! I am just a wikipage reader now, but as I get better into the subjects, I will hopefully contribute as well.  

See you there!