ArduPilot/ardupilot_wiki #2725: Update pixracer serial port map

kd0aij, 2020-05-07:
Below is the mapping of serial ports to uarts given at https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/Pixracer/README.md

It appears to be correct, expect for labeling serial5 as spare, when it is actually connected to the ESP8266 port.

Do we want to update the corresponding wiki page at
https://github.com/ArduPilot/ardupilot_wiki/blame/master/common/source/docs/common-pixracer-serial-names.rst
or just provide links to the readme files in the source?

## UART Mapping

 - SERIAL0 -> USB
 - SERIAL1 -> UART2 (Telem1)
 - SERIAL2 -> UART3 (Telem2)
 - SERIAL3 -> UART4 (GPS)
 - SERIAL4 -> UART8 (GPS2, FrSky Telem)
 - SERIAL5 -> UART1 (spare)
 - SERIAL6 -> UART7 (spare, debug port)

Hwurzburg, 2020-05-07:
PR #2703 in and awaiting approval for merge...

kd0aij, 2020-05-07:
thanks, I failed to search for other issues or PRs

closed 2020-05-07 by kd0aij