ArduPilot/ardupilot_wiki #2729: Benewake TFmini / TFmini Plus lidar

patrickpoirier51, 2020-05-08:
[Benewake TFmini / TFmini Plus lidar](https://ardupilot.org/copter/docs/common-benewake-tfmini-lidar.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I think is paragraph is misleading
_RNGFND1_TYPE = 20 also works for TFminiPlus via its UART connection, if connected via I2C to the autopilot, use RNGFND1_TYPE = 25 (Benewake TFmini/Plus-I2C) and also set RNGFND1_ADDR = 16 (I2C Address of lidar in decimal). The sensor’s default I2C address is 0x10 hexademical which is 16 in decimal._

Because on I2C the TFMINI and TFMINIPLUS are not register compatible.

As explained here: https://github.com/budryerson/TFMini-Plus-I2C
you can read:
The Benewake TFMini-Plus is a unique product, and it is not an enhanced version of the Benewake TFMini. It has its own command and data structures. This library is not compatible with the TFMini product.

and tested here:
https://discuss.ardupilot.org/t/range-finder-greyed-out/55918/20

@lucasdemarchi  comments please

Hwurzburg, 2020-05-08:
Can you state exactly what is incorrect about the paragraph? the wiki currently does not indicate that I2C connection of the TFMini is possible....only the TFMiniPlus...prior the the paragraph it was talking about UART connections with both support...if TFmini is supported in ArduPilot via I2C to the FC, we need to state how that is configured...

patrickpoirier51, 2020-05-08:
Oh chute , I just saw that the misleading word is indeed the sensor definition
RNGFND1_TYPE = 25 (Benewake TFmini/Plus-I2C)
 reading the above makes the user think that both the TFMini and TFMini Plus can handle I2C
The fact is the code for TYPE= 25 apply to TFMini Plus exclusively

Hwurzburg, 2020-05-08:
the problem is in the meta data for the param.....type 25 is labeled as such.....if it can be changed, we can reflect that in the wiki....

Hwurzburg, 2020-05-17:
@rmackay9 what do you think...

Hwurzburg, 2020-05-20:
okay, the meta data label for type 25 will be changed and I will change the wiki...will close this when thats done

closed 2020-06-21 by Hwurzburg